maplerainresearch/MRR_ESPE #2: MRR ESPE with Trinamic drivers in UART mode

vivian-ng, 2019-12-25:
I received an email asking about TMC2209 drivers in UART mode. I thought it best to move the conversation to a more public space so that the collective wisdom of the open source community can be harnessed.

First, @luc-github has managed to use TMC2208 in standalone (aka "dumb") mode which does not use UART at all. Basically, the driver is configured using jumper pins much like normal "dumb" drivers like A4988.

For the MRR ESPE, the issues with getting UART mode working are:
1. **Native ESP32 pins**. Unlike the MRR ESPA, the MRR ESPE has more unused native pins since I2S is being used for stepper control and PWM. So by default, the MRR ESPE has 5 unused pins. If the LCD controller is not used, this frees another 6 pins for a total of 11 pins. Which means 5 axes (5 stepper drivers) can be supported as long as the serial ports for them can be configured.
2. **Hardware serial**. While the ESP32 has 3 sets of hardware serial, I have seen a lot of articles regarding getting all 3 to work. Most can get 2 sets working. One set will be used for serial terminal (USB to UART) already; that leaves only one other set of hardware serial for TMC driver in UART mode.
3. **Software serial**. The lack of hardware serial is usually overcome with software serial. But software serial support for ESP32 on the TMCStepper library is still being worked on. Unless someone is able to work on the software development for this, software serial will not be an option at all. But if someone manages to develop support for software serial on the ESP32 in the TMCStepper library, the MRR ESPE can theoretically use TMC drivers in UART mode (if the LCD controller is not used).

Consolidating:
- Trinamic drivers in standalone mode works.
- Trinamic drivers in SPI mode works. If you need silent operations, TMC2130 drivers can be used. For higher currents, TMC5160 should work too.
- Trinamic drivers (TMC2208 and TMC2209) in UART mode has the limitations as outlined above.

vivian-ng, 2019-12-25:
For use of Trinamic drivers in UART mode on the MRR ESPA (which does not use I2S and therefore has fewer available pins), see this [issue](https://github.com/maplerainresearch/MRR_ESPA/issues/2).

Please feel free to join in the conversation.

luc-github, 2019-12-25:
There no software serial on esp32 because any native pin can be repurposed as hardware serial
Do you mean use expander pins for software serial? I do not think there is any plan for this

vivian-ng, 2020-01-29:
I was not very informed about the TMC2209 stepper drivers. TMC2208 has a fixed slave address, which is 0, and that means only one driver can be on the same UART port since there isn't any addressing capability. But TMC2209 allows configuration of up to four different slave addresses (0 to 3). This means up to four stepper drivers can actually share the same UART port. With @luc-github's help, I managed to get two TMC2209 drivers working on the same hardware serial on the MRR ESPA.

See this video for a quick look at how to set this up for the MRR ESPA.
[image]

See this [blog post](https://teck78.blogspot.com/2020/01/setting-up-tmc2209-in-uart-mode-on-mrr.html) for more details.

For the MRR ESPE, the process should basically be the same if up to four TMC2209 stepper drivers are used. To use TMC2209 stepper driver on the 5th motor, a second UART port will be needed. This means adding something like
```
Serial2.begin(115200, SERIAL_8N1, 2, 4);
```
somewhere in Marlin to set up `Serial2`. This has yet to be tested though, because I only have two TMC2209 stepper drivers at the moment, and a new order is likely to be delayed since Aliexpress is on extended Chinese New Year break due to the coronavirus.

vivian-ng, 2020-01-31:
I did a test print using TMC2209 on X and Y axes, it seems to be okay.
[image]