maplerainresearch/MRR_ESPE #3: Problem initializing LCD controllers

vivian-ng, 2020-02-10:
A recent phenomenon has appeared and I have not managed to find time to fully track down the problem. Out of the blue, I have not been able to get LCD controllers to work with the MRR ESPE. they were working fine, then when I power on the printer the next day, I get a blank LCD screen (only backlight) and the control board does not boot up.

The issue started with CR10 stock displays, but I recently experienced this on my RepRap full graphics smart controllers (both of them, on two different MRR ESPE boards) too. All of these controllers were working fine with their respective MRR ESPE boards previously, so I do not think it is a hardware issue. Still, if anyone experiences such an issue, please contribute information here and see how we can collectively work out what the problem is.

luc-github, 2020-02-10:
Well my CR10 died in that way 
why do you think it is not hardware issue,  does reflasing FW solve issue ?

vivian-ng, 2020-02-10:
@luc-github Okay, I am in the realm of guessing now... but my suspicion is the BTN_EN1 pin, which is connected to GPIO0. My guess now is that if, somehow, the BTN_EN1 is pulled to GND by the LCD controller, it is causing the board to do into download (flash) mode, which may explain the blank screen and the board failing to boot properly.

The "clone" CR10 stock display I had just miraculously worked again. I plugged it into one of my boards, blank screen. Then, I turned the knob (rotary encoder) a bit, connected the LCD again, and suddenly, the LCD came alive.

My RepRap full graphics smart controllers are also working now. On the exact same board that was giving me problems last night.

Maybe you can try turning the knob of your LCD controller and try again? I have a print going on my other printer, so when that one is done, I will test this theory out too.

luc-github, 2020-02-10:
ho nice - I am still away from home and I do not have currently espe board neither cr10 lcd on hand - so I won't be able to test before a while

vivian-ng, 2020-02-10:
LCD is working on my stock Ender-3 display too. I guess it is really an issue with BTN_EN1 being pulled to GND (somehow) when powering up.

If anyone turns on their printer and the LCD shows a blank screen, and keeps beeping (you will also notice that your hotend and heated bed and every fan is being activated), it is likely this issue. Quickly power off the printer. Then turn the LCD knob a bit. Then power on again. That should solve the problem if it is really an issue with BTN_EN1 (aka GPIO0) being pulled unintentionally to GND during boot up.

closed 2020-02-10 by vivian-ng

luc-github, 2020-02-10:
could this lead to fry the IO expanders like I got,  if not powered off quickly enough?

vivian-ng, 2020-02-10:
@luc-github 

> could this lead to fry the IO expanders like I got, if not powered off quickly enough?

This should not happen, since this is exactly the scenario when flashing firmware via USB. I will also add a note that the firmware should not be flashed via USB when the heated bed and hotend are connected. If firmware must be flashed in such situations, the 5V_SEL jumper should be set to VUSB, and the main power supply to board and heated bed should be turned off.