maplerainresearch/MRR_ESPE #6: Abnormal temperature display, 23 ℃ without inserting temperature sensor

dream111119, 2021-01-22:
Hello, I have trouble using MRR espe today. I use a 100k thermistor. Configure. H chooses to use a temperature sensor. The serial number is 1. But when I pulled out the temperature sensor, the temperature was 23 degrees. I checked the resistance configuration and so on, and there was no problem. I want to know the reason, and then how to solve it? thank you

vivian-ng, 2021-01-22:
@dream111119 I have this issue too. It has to do with the ESP32's ADC and we have not found a way to get it to display any value lower than 20+ degC. You can see the discussion here on [this thread](https://github.com/simon-jouet/ESP32Controller/issues/10). So there will be a "baseline" value of 20+ degC even if the thermistor is not connected. Usually, the hotend and heated bed should heat up to a temperature higher than this "baseline" temperature within the stipulated thermal runaway time.

Try heating your hotend to 200 degC and heated bed to 50 degC to see if there is any problem when the temperature goes above 23 degC.

dream111119, 2021-01-22:
Thank you. I haven't found a way to check the data of esp32-adc. When it is lower than 23 degrees, the output ADC value is the same as when it is 23 degrees. I plan to test the ADC separately. It is estimated that there will be no harvest. It's very hard