maplerainresearch/MRR_ESPE #7: When IO0 is connected to CR10 display screen, the reset will be invalid

dream111119, 2021-01-28:
Hello, I have trouble using IO0. I can't use the reset function. After clicking reset, I didn't enter the main program. Is there any way for IO0?

luc-github, 2021-01-28:
if IO0 is connected to ground at start, the board go to flash mode (https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#gpio0)  - if you use IO0 be sure it is not connected to GND at start or your board will not start

vivian-ng, 2021-01-28:
@dream111119 Please see if this [related issue](https://github.com/maplerainresearch/MRR_ESPE/issues/3) helps. Sometimes, the control knob on the LCD pulls the BTN_ENC1 to GND, which causes a problem for the ESP32.

If the reset button on the main board works, it could be an issue with the LCD controller itself. Make sure the connections from the LCD are properly connected to their required connectors on the main board.

vivian-ng, 2021-01-31:
@dream111119 If the issue is resolved, please share with everyone what was done to resolve the issue so that others with such an issue next time can try.

vivian-ng, 2021-03-22:
Closed due to no further response.

closed 2021-03-22 by vivian-ng