maplerainresearch/MRR_ESPE #8: Second, the problem of serial port open failure

dream111119, 2021-01-29:
Hello, I want to know how to open the second serial port? The best is serial port 1, I need to use a serial port touch screen here, thank you

dream111119, 2021-01-29:
After I opened it, the compilation passed, but actually it was useless. Do you want to know that it was like this? It's very painful

vivian-ng, 2021-01-29:
@dream111119 Please provide more information about your exact setup so that we can better understand the issue. For example, what board are you using? What LCD controller? What are the pins you are trying to use? Are you using TMC stepper drivers in UART mode? What have you actually defined in Configuration.h and Configuration_adv.h, and have you made any changes to the pins file?

dream111119, 2021-01-29:
First: I use bigtree tft35, which is a serial screen
Second: the screen works the same as online printing. I use serial port 1 to connect it.
Third: the screen itself is normal
Question: I opened serial in configuration. h SERIAL_PORT_2(1), and in pins_ MRR_ESPE.h is turned on in HARDWARE_SERIAL1_TX and HARDWARE_SERIAL1_RX, found unable to use

dream111119, 2021-01-29:
The click driver is A4988. Normal operation

luc-github, 2021-01-29:
Unfortunalty Marlin allows only 2 Serial_ports
For ESP32
One is for USB or any serial output but only one can be used at once USB is first if anything is connected if I am not wrong
One is for WIFI communication 

if you can access to RX1/TX1 pins of board which are not on serial header - and configure SERIAL_PORT_2 1 - you will loose WiFi feature

On my MRR_ESPE + BTT TFT I use the Serial header  for TFT,  BTT TFT has a feature that disconnect serial of TFT and so allow to switch to USB serial

vivian-ng, 2021-01-29:
@dream111119 
Try connecting the TFT35's TFT connector to AUX1 on MRR ESPE.
TFT -> MRR ESPE
5V -> 5V
GND -> GND
RX -> TX
TX -> RX

For RST on the TFT, you can connect to the RST pin on the MRR ESPE's LCD connector.

In `Configuration.h` this setting should allow you to use the AUX1 connector of MRR ESPE to communicate with TFT35 via serial. It will share the same serial as the microUSB connector on the board so you cannot use TFT35 if you have the MRR ESPE connected to a computer (like PC or Raspberry Pi) via USB.

```
#define SERIAL_PORT 0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
//#define SERIAL_PORT_2 -1

/**
 * This setting determines the communication speed of the printer.
 *
 * 250000 works in most cases, but you might try a lower speed if
 * you commonly experience drop-outs during host printing.
 * You may try up to 1000000 to speed up SD file transfer.
 *
 * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
 */
#define BAUDRATE 115200
```

luc-github, 2021-01-29:
@vivian-ng in latest Marlin you must now uncomment manually `#define SERIAL_PORT_2 -1` to get wifi working

vivian-ng, 2021-01-29:
> @vivian-ng in latest Marlin you must now uncomment manually `#define SERIAL_PORT_2 -1` to get wifi working

Thanks @luc-github for pointing this out.

dream111119, 2021-01-29:
Thank you very much, but I have two more questions.
First: for esp32, if I connect BTT TFT, can I open serial port 1 or serial port 2? Or reuse to other pins
Second: #define SERIAL_PORT_2 -1  have to be defined like this?

vivian-ng, 2021-01-29:
@dream111119 
You should be able to set specific pins for Serial1 by:
```
#define HARDWARE_SERIAL1_RX                 21
#define HARDWARE_SERIAL1_TX                 22
```
in `Configuration.h` or `Configuration_adv.h`. Just put these lines anywhere, and replace 21 and 22 with the pins you want to use. You can then use Serial1.

But like what @luc-github said, if you, if you do
```
#define SERIAL_PORT_2 1 
```
them you will not be able to use Wifi. So it is either TFT35 in serial mode, or Wifi. You may need to choose if you want to use Serial1.

But if you are using MRR ESPE and BTT TFT35, you should be able to use EXP3 connector on BTT TFT35 (connect to LCD connector on MRR ESPE) for the controller in LCD12864 emulator mode. Then, in `Configuration.h`, just:
```
#define CR10_STOCKDISPLAY
```
will do. Personally, I use Ender-3 stock display with MRR ESPE using the EXP3 connector. I have also tried with BTT TFT35 E3 V3.0 using EXP3 too. It works if you use the LCD12864 emulator mode.

vivian-ng, 2021-01-31:
@dream111119 If the issue is resolved, please share with everyone what was done to resolve the issue so that others with such an issue next time can try.

vivian-ng, 2021-03-22:
Closed due to no further feedback.

closed 2021-03-22 by vivian-ng