maplerainresearch/MRR_ESPE #9: TMC2130 cannot connect

dream111119, 2021-01-31:
Hello, thank you for your guidance last time. I solved the problem of TFT screen. But when I test tmc2130, I find that it can't be used. I send the M122 command, but the return is not OK. I use bigtree tech tmc2130 v3.0
1. I'm in pins_ MRR_ESP32 Turn on the chip selection pin of motor drive , and the jumper cap on the hardware is connected correctly
2. I set it in configuration. h   #define X_DRIVER_TYPE  TMC2130

dream111119, 2021-01-31:
Question: first, do I have any missing settings?
Second, will tmc2130 use the same SPI as SD card?
Third, can tmc5160 be used normally?

dream111119, 2021-01-31:
@vivian-ng

vivian-ng, 2021-01-31:
@dream111119 Please provide more information, such as the actual output from M122, and the relevant portions of Configuration.h and Configuration_adv.h so that people can better understand what the issue is.

dream111119, 2021-01-31:
M122 output:   Testing X connection... Error: All LOW
configuration.h: #define X_DRIVER_TYPE  TMC2130
configuration_adv.h:No change
pins_MRR_RSPE.h:  open #define X_CS_PIN   21
I only use it on the x-axis

dream111119, 2021-01-31:
@vivian-ng

vivian-ng, 2021-01-31:
For a start, please see the [manual](https://github.com/maplerainresearch/MRR_ESPE/blob/master/MRR%20ESPE%20manual.pdf) which contains basic information about the jumpers and wiring that is required for using TMC2130. It also contains information about how to set the CS pins in the configuration files.

The TMC2130 will share the SPI with the microSD card. Please also note this issue [here](https://github.com/maplerainresearch/MRR_ESPE/issues/1#issuecomment-574941070).

dream111119, 2021-01-31:
I did it according to the manual and also referred to the manual of motor drive, but I still couldn't do it. It's embarrassing. M122 is still a connection failure

dream111119, 2021-01-31:
@vivian-ng

vivian-ng, 2021-01-31:
@dream111119 Is the board and motor powered by VIN (12-24V)? If you are using only USB power (5V), the stepper drivers are not powered and you get connection error.

vivian-ng, 2021-01-31:
@dream111119 Also, I think the place to define the CS pins should be in `Configuration_adv.h` now. In addition to the steps in the [MRR ESPE manual](https://github.com/maplerainresearch/MRR_ESPE/blob/master/MRR%20ESPE%20manual.pdf), please also check the [Marlin documentation](https://marlinfw.org/docs/hardware/tmc_drivers.html) on how to enable TMC2130.

The MRR ESPx series of boards are experimental prototypes and assumes that the user has prior knowledge working with other 3D printer boards (like RAMPS) and Marlin. Otherwise, the learning curve can be quite steep. It is also good to take things in stages. Like first, get the board up and running with a basic setup (single hotend, no heated bed, one fan, A4988 stepper drivers). Once you can get it running a basic setup, then slowly add stuff (like LCD controller, TMC stepper drivers, more controllable fans, NeoPixels, etc.) This was basically my approach too when working on this board.

vivian-ng, 2021-03-22:
Closed due to no further response.

closed 2021-03-22 by vivian-ng