maplerainresearch/MRR_ESPE #10: Thermistor temperature is not correct

RobertoRmz, 2021-03-22:
I have problems with the Extruder Ntc 3950 100k Thermistor, it s**hows a different temperature to that of a digital thermometer**, the following image shows some tests that I did changing the sensors available in Marlin and the results that I obtained. I did this as well by changing the MOTHERBOARD [MMR ESPA and MMR ESPE] option in Marlin as they drive the same thermistor pin on the ESP32. 

These tests were carried out with a digital thermometer that read 25.5 ° C, option 332 was the option that was closest to the thermometer temperature of 4 degrees difference. 

[image]

However, another problem that I found with option 332 was that it no longer drops more than 29.5 ° C, even though I place a fan, an ice no longer goes down, it always stays at 29.5 ° C.

Also change values of the following Marlin options to see if there was a different value in temperature, but nothing happened:
1- #define MAX_REDUNDANT_TEMP_SENSOR_DIFF 10
2- #define TEMP_HYSTERESIS 3

I would appreciate any advice in which you could help me so that the thermistor can measure the temperature correctly.

vivian-ng, 2021-03-22:
@RobertoRmz Please see this [issue](https://github.com/maplerainresearch/MRR_ESPE/issues/6). It has to do with the ESP32's ADC. Try thermistor option 1, then heat the hotend to a slightly higher temperature (100 degC?) to see if there is a significant difference between thermistor reading and digital thermometer reading. There is actually no real solution to fix this right now for idle (room) temperature, the only solution is to see if the temperature difference at operating temperature (around 200 degC for hotend, 60+ degC for heated bed) affects printing.

RobertoRmz, 2021-03-22:
@vivian-ng thank you very much for the help, if I read [Known issue Abnormal temperature display, 23 ℃ without inserting temperature sensor] but I had not understood, but you already made it clear to me by telling me that there is no real solution to fix this right now for ambient temperature. 

i did the test as you said i heated to 200 degC and using option 1 on marlin sensors and it worked fine. 

I do not recommend measuring with a digital thermometer because there is a lot of temperature difference. What I did was connect the Ramps 1.4 to use only the Thermistor and place it in the Aluminum Heater Block where the ESP32 thermistor is, send it to heat to 200 degC and the results were the same, again thank you very much for the help vivian-ng, regards.