maplerainresearch/MRR_ESPE #11: Homing Problems

RobertoRmz, 2021-03-27:
Whenever I do Homing either on the X, Y or Homing on all axes, it is always the same problem.
**First the Z axis motor advances**, then the X-Y motor advances a little and it marks this error. 

[image]

So that this error does not appear when I do Homing in the X or Y axis, the Z axis motor moves and then the X-Y, but if I quickly press the Limit Switches it does not mark that error.

It cannot be a Hardware error (Bad connections made) because when sending M119 to detect the state of the Limit Switches, it recognizes them correctly: 

[image]

I think that some more configuration must be missing so that I can do the Homing correctly, in configuration.h in El Marlin 2.0, I have the following configuration: 

[image]

Probe enabling and disabling the ENDSTOPPULLUPS and I keep getting the same result 

[image]

[image]

This configuration works perfectly for me but using Arduino Mega with the Ramps 1.4 in the Marlin 1.1.X using the Core XY configuration, but in the ESP32 it does not work for me, there is another place where I can configure the Limit Switches so that it can work correctly, I would appreciate help a lot.

vivian-ng, 2021-03-28:
@RobertoRmz I am not really sure what the issue is. Are you using Marlin 1.1.x or Marlin 2.0.x? There are many differences in the configuration files between 1.1.x and 2.0.x, so a configuration that works for 1.1.x will likely not work for 2.0.x. Also, ESP32 needs to use Marlin 2.0.x, and even within 2.0.x, the configuration files have changed significantly during the past year. Some features have been renamed.

To make sure that this is not a Marlin issue, please try using an older release of Marlin 2.0.x from the releases [here](https://github.com/MarlinFirmware/Marlin/releases). I think 2.0.3 should work. You will need to work through the configuration files (Configuration.h and Configuration_Adv.h) to edit them for your setup because these two files have changed significantly over the past year. If 2.0.3 works, then try the next newer release to see if it works. Eventually, you should come to a release version that doesn't work, and that should provide information on the changes in configuration files (and firmware) that could have caused some issues.

Another troubleshooting method that I always use (besides falling back to older releases) is to start with a basic setup. A4988 stepper drivers, cartesian printer, physical endstops, etc. If the most basic setup works, then I slowly add features (s curve acceleration, TMC drivers, corexy, etc.) one by one to see which could be the feature that is giving a problem.

Also, try with Marlin 2.0.x on your RAMPS 1.4 board. If your setup can work in Marlin 1.1.x but not in 2.0.x, you may want to try searching around the Marlin repo to see if someone else has experienced the same issue. Cannot really compare Marlin 1.1.x running on RAMPS 1.4 with Marlin 2.0.x on ESP32. Too many factors have changed for proper comparison.

RobertoRmz, 2021-03-30:
I was using Marlin 1.1.X because I was using Ramps 1.4, but I have been reading about the newer 32-bit printers and they have mentioned that they have more processing power and therefore better print quality.

I am configuring ESP32 with Marlin 2.0.X in Visual Studio Code with PlatformIO and testing with Driver A4988.

I managed to solve the problem, the only thing I did was change the ESP32 for a new one and the Homing worked correctly, with Limit Switch on the x-axis and y-axis.

  In the Z Axis I put an inductive sensor at the beginning I struggled a lot because I got this error "_**#error" Z_SAFE_HOMING is recommended when homing with a probe. Enable it or comment out this line to continue.**_ ", I solved this by uncommenting **#define Z_SAFE_HOMING.** . @vivian-ng Thank you very much again for your help, many regards.

closed 2021-03-30 by vivian-ng