maplerainresearch/MRR_ESPE #12: Hotend temperature is not stable

RobertoRmz, 2021-04-02:
Make prints at 205, 210,215 and 225 deg, However, when printing the same problem always happens. The HotEnd temperature varies widely **± 15 deg**. It must be the PID values, these values were obtained from Pronterface's AutoTune, the values with which I made the last print were the following: 

**_a)	Kp = 10.59
b)	Ki = 1.70
c)	Kd = 16.52_**

I show two temperature graphs from when I printed in Octoprint, it is possible to observe how there are very high temperature peaks: 

[image]

[image]

I have been testing various values in the PID but still with no success. Some advice to be able to determine the correct values Kp, Ki and Kd so that the temperature of the hotend can be kept stable like the hot bed.

vivian-ng, 2021-04-02:
`M303 C10 E0 S200`
Replace the `200` in `S200` with whatever is your usual hotend printing temperature. This should print out the PID values after going through 10 cycles.

`M303 C10 E0 S200 U1`
if you want to apply the PID settings immediately after the autotune is completed.

Remember to use `M500` to save the PID values. Else they disappear upon reset/power down.
[PID autotune](https://marlinfw.org/docs/gcode/M303.html)

Another thing to note is that temperature reading values can be easily affected by EM noise. Should not be too much of an issue if you used the right capacitors for filtering.

Trying looking at the [issues here](https://github.com/simon-jouet/ESP32Controller/issues) to see if anything looks like what you are experiencing. That design is the original base design for control boards based on the ESP32.

vivian-ng, 2021-04-02:
Closing as this is an issue arising from a breadboard configuration and not the MRR ESPE board.

closed 2021-04-02 by vivian-ng