maplerainresearch/MRR_ESPE #13: LCD screen problem

RobertoRmz, 2021-04-02:
After assembling the printer thanks to the diagram provided by maplerainresearch, although I made some changes because it did not have many of the components of the diagram, for example to turn on the hot bed, hotend and fan, use Mosfet Irf3205. 

[image]

It worked and I was able to print the Calibration Cube 20x20 mm, more tests need to be done to calibrate it so that it can print perfectly. 

I need to connect the screen. Connect 12864 lcd full graphic smart controller that already has SD module, present the following problem: 

[image]

[image]

The first thing I did was investigate the pins on the screen (EXP1 and EXP2) to know where to connect it on the ESP32 

LCD screen: https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
Pin Diagram: https://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf

[image]

Connect the display pins EXP1, EXP2 according to the pins that appear in pins_MRR_ESPE. 

[image]

According to this information, make a diagram of how to connect the screen with the ESP32, I did it on a different breadboard than the one in the photo I put at the beginning. 

[image]

As I show in the diagram there are 4 pins in EXP2 that I did not know where to connect it to the ESP32 **_"It may be that that is the error"._** 

In the Marlin configuration enable the following 2 options “**#define SDSUPPORT** and in CONTROLLER TYPE: Graphical 128x64 (DOGM) **#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER**. I would appreciate some advice so that the screen can work properly.

vivian-ng, 2021-04-02:
@RobertoRmz 
See schematic [here](https://github.com/maplerainresearch/MRR_ESPE/tree/master/adapters/LCD_adapter).

Info on how to use LCD12864 was posted [here](https://github.com/maplerainresearch/MRR_ESPE/issues/1#issuecomment-567368605) and [here](https://teck78.blogspot.com/2019/12/the-lcd-connector-on-mrr-espe-which-is.html). You can always search the issues to see if something has already been written instead of opening a new issue.

vivian-ng, 2021-04-02:
@RobertoRmz 
Also, please note that this board was not designed to be used on a breadboard. It is hard to provide any help unless you are using a proper PCB, since a breadboard setup can have other types of errors like loose connections, broken wires, and EMI that is different from what the PCB version has.

vivian-ng, 2021-04-02:
Closing as this is an issue arising from a breadboard configuration and not the MRR ESPE board.

closed 2021-04-02 by vivian-ng