maplerainresearch/MRR_ESPE #14: Generic I2S questions

petervanderwalt, 2021-11-10:
Hi!.  I am experimenting with TMC5161 drivers, on a breadboard setup running an ESP32 with the I2S expanders.  

I have just about everything else working, i can control dumb drivers just fine, but I don't seem to be getting the CS pin to switch for spi transactions to the TMCs.  

Verified with a scope, activity on SCK and MOSI, but CS just stays high. Do you know if 

a) its possible to use on of the I2S expansion pins as a CS pin for the Trinamics (no spare GPIOs on the ESP32, and I know Esp32_Grbl can use the I2S pins as CS pins for TMC and 

b) any pointers on making it work?

M122 reports ALL HIGH so I know its a comms issue, and not seeing CS go low at the start of the transaction makes me suspicious of whether I can use I2S CS pins in Marlin?

Also, can I run more than 32 bits? For example, can I run 6* 595s for 48 extra Outputs?

vivian-ng, 2021-11-10:
I am afraid I have never tried using TMC5161 drivers before so I am not able to provide any help on this. Hopefully someone else who has tried this wanders here and sees these questions. Will leave this issue open in case someone else can provide more insights.

petervanderwalt, 2021-11-11:
Thanks for getting back to me (: 

Shout if you'd like a set I am sure I can arrange some for you (: